dotnetprojects/DotNetSiemensPLCToolBoxLibrary #32: Block TIMESTAMPs not Correct

proemmer, 2015-04-09:
Hi ,

in the File LibNoDaveConnectionLibrary\DataTypes\Projectfolders\Step7V5\BlocksOfflineFolder.cs there are some Bugs in the GetBlockBytes Method.

1. Based on my tests, in the rows with the follwoing subblktypes are the correct typestamps per blocktype:
12 - FC
8   - OB
14 - FB
13 - SFC
15 - SFB
10 - DB

2. The calculation of the timestamp is sometimes not correct because of the Encoding:
    My suggestion would be, to Change this:
```
var ts1 = (string)row["TIMESTAMP1"];
Helper.GetDT((byte)ts1[0], (byte)ts1[1], (byte)ts1[2], (byte)ts1[3], (byte)ts1[4], (byte)0);
```
to that:
```
//use Windows-1252 code page to get correct time 
//because dBaseConverter uses this codepage for strings
var bytes = Encoding.GetEncoding("Windows-1252").GetBytes((string)row["TIMESTAMP1"]);  
Helper.GetDt((byte) bytes[0], (byte) bytes[1], (byte) bytes[2], (byte) bytes[3], (byte) bytes[4], (byte) bytes[5]);
```

Best regards,
Benji

jogibear9988, 2015-04-09:
could you create a pull req?

proemmer, 2015-04-09:
I have changed the file and commited, hope this is ok, it's my first time on GitHub.

jogibear9988, 2015-04-09:
Now you can create a Pull Req wich I can accept, if you don't know how I can do it...

https://help.github.com/articles/using-pull-requests/

closed 2015-04-09 by jogibear9988

proemmer, 2015-04-09:
OK, Done.
Thanks for the link.

One other question, have you ever tried to extract pdiag data from the dbf files? Especially the data for the motion lines.  I think the data are in PDIAG.DBF and the Texts are in textable.dbf but i find no sense in the data.

jogibear9988, 2015-04-09:
No, I never used PDIAG! It was a huge part of work to find sense in the DBF File Data...

jogibear9988, 2015-04-09:
What Infos do you need from PDIAG?

proemmer, 2015-04-09:
1. I need the motion data ->

proemmer, 2015-04-09:
2: I need the Alarm Data: When I read the Alarmlist from the PLC  I would convert the alarm number to the text.

jogibear9988, 2015-04-09:
Ok, never tried this. I found out of other DBF Files, by for example change a Bit address from 1 to 2 and see what changes in the DBF... And after many, many attempts I found out what you see in this lib.

proemmer, 2015-04-09:
ok, thank's  I will keep trying.