dotnetprojects/DotNetSiemensPLCToolBoxLibrary #40: Issue by loading projects with s7 programs without a PLC

impoetk, 2015-11-21:
I have found that an exception is thrown when opening projects with offline foldes that have no CPU as parent.

im not sure why, but the exception is thrown in the linq expression in the Step7ProjectV5.cs LoadProject Method

             foreach(var dp in DPlist)
            {
                var dpf = DPFolders.FirstOrDefault(x =>  x.IdTobjId.Any(y => y == dp.TobjID));
                if ( dpf != null)
                {
                    if (dpf.TobjId == null) dpf.TobjId = new List<int>();
                    dpf.TobjId.Add(dp.addr);
                }
            }

to fix it i have modified the linq to the following

                var dpf = DPFolders.FirstOrDefault(x => x.IdTobjId != null && x.IdTobjId.Any(y => y == dp.TobjID));

jogibear9988, 2015-11-21:
could you create a pull req?

jogibear9988, 2016-01-31:
closed via #41

closed 2016-01-31 by jogibear9988