dotnetprojects/DotNetSiemensPLCToolBoxLibrary #56: Failed to parse S7P file

DJWassink, 2016-08-02:
I am trying to open a S7P file which worked fine on a older version (like 2 years old or something) but one the recent versions it keeps crashing. 
The rows it keeps on crashing looks like this:

```
-		rows	{string[12]}	string[]
		[0]	"\r"	string
		[1]	"VAR_TEMP\r"	string
		[2]	"  OB86_EV_CLASS : BYTE ;\t//16#38/39 Event class 3\r"	string
		[3]	"  OB86_FLT_ID : BYTE ;\t//16#C1/C4/C5, Fault identifcation code\r"	string
		[4]	"  OB86_PRIORITY : BYTE ;\t//Priority of OB Execution\r"	string
		[5]	"  OB86_OB_NUMBR : BYTE ;\t//86 (Organization block 86, OB86)\r"	string
		[6]	"  OB86_RESERVED_1 : BYTE ;\t//Reserved for system\r"	string
		[7]	"  OB86_RESERVED_2 : BYTE ;\t//Reserved for system\r"	string
		[8]	"  OB86_MDL_ADDR : WORD ;\t//Base address of IM module in rack with fault\r"	string
		[9]	"  OB86_RACKS_FLTD : ARRAY  [0 .. 31 ] OF BOOL ;\t//Racks in fault\r"	string
		[10]	"  OB86_DATE_TIME : DATE_AND_TIME ;\t//Date and time OB86 started\r"	string
		[11]	"END_VAR\r"	string
```

As you can see line 9 is a array with some comment behind it saying "//Racks in fault"
In the Parameter.cs file (PLCs\S7_xxx\MC7\Parameter.cs) it appends line 10 to line 9 resulting in the following line: `OB86_RACKS_FLTD : ARRAY  [0 .. 31 ] OF BOOL ;\t OB86_DATE_TIME : DATE_AND_TIME ;\t//Date and time OB86 started//Racks in fault`

This results in the following `tmpType` ` ARRAY  [0 .. 31 ] OF BOOL 	 OB86_DATE_TIME : DATE_AND_TIME 	`

Since there isn't a S7DataRowType like this it eventually fails to parse this row.

DJWassink, 2016-08-02:
Simply changing the checkout on line 215 from `\t\r` to `\t` solved my issue. Though I am not sure if this is the correct way to fix this edge case and if it will introduce other bugs.

https://github.com/DJWassink/DotNetSiemensPLCToolBoxLibrary/commit/0003e082431f67aa394cc70e50629141eb1e7cab

jogibear9988, 2016-08-18:
For me its also not sure. will try with a few projects in our company and fix it

jogibear9988, 2016-08-20:
Maybe you could send me a Project File with your Block in it? So I could create a Test and will not reintroduce the Bug?

DJWassink, 2016-08-20:
I am currently on a holiday and don't have access to my stuff. I am flying
back soon and will get you a file asap. Thanks.

DJWassink, 2016-08-23:
Here you go, It includes the ODB86 which is the conflicting block. Good luck and thanks again!

[odb86.zip](https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/files/431984/odb86.zip)

jogibear9988, 2017-10-27:
this block works in newest version

closed 2017-10-27 by jogibear9988