dotnetprojects/DotNetSiemensPLCToolBoxLibrary #61: Language support

impoetk, 2016-09-12:
Hi, I was wondering if its possible to parse different languages when reading symbols and datablocks, i guess when user is changing the language in the simatic manager the language information is taken from somewhere and written in the corresponding database files.

do you have any information where this language file is? or how is it done internally?

jogibear9988, 2017-10-27:
as i know, the project has only one language

impoetk, 2017-12-20:
in simatic manager v5.5 it is possible to change the language for almost everything inside, symbol comments, datablock entries comments, network titles and comments, etc, and the language is switchable inside the simatic manager, so when the language is changed this library takes the current language selected, i have projects in german-english, german-spanish and for some HMI tools we have to take the comments on each language, so the need of switching the language in simatic manager is cumbersome and prone to errors, so i was wondering if its possible to obtain such texts from within the project file itself.

jogibear9988, 2017-12-20:
I really don't know. Maybe you can look for yourself in the Database Files? I've not really time to work on this, but I would help you if you can do!

closed 2026-10-21 by jogibear9988