dotnetprojects/DotNetSiemensPLCToolBoxLibrary #62: Strange Block Upload behavior on S7-400

StefanHasensperling, 2016-10-03:
Recently y did some tests with the "PLCGetBlockInMC7" on S7-400 PLCs. The issue that i will describe here is, i think, not an bug in this library (or libnodave), but rather an bug in either the S7-Firmware or something similar. Maybe somebody knows something about this issue, or it will help somebody encountering the same issue.

**The solution was easy:** I retry the upload once. This solved the issue for me

**First off some information about the software involved:**
- DotNetSiemensToolboxLib: 2.0.127 from 12.09.2016
- S7: 416-2 DP 6ES7 416-3XL00-AB0
- S7-400 Firmware: 3.1
- S7-400: Hardware: I think it was 8, but i might be wrong here
-CP: 6GK7 443-1EX11-0EX0 V1.1
-Connected via ISO_over_TCP

All blocks that i was testing, where all definitely existing in the CPU.

**Now to the issue i encountered.** 
When you connect to the PLC always the first block that you try to upload via PLCGetBlockInMC7 will fail with 0x8702 "address invalid". It does not matter what block you choose, always the first one will fail  immediately (no timeout or something). When you try the second block, even if it is the same, it works without any problem. 

I tested this, and could reproduce this behavior consistently. Always the first attempt to upload an block will fail, and the second one will succeed. It will fail The strange thing is, that it fails witch 0x8702 Address invalid, as if you are requesting some non supported block or something. 

I traced the error and found that the library (and Libnodave) are behaving correctly as fat as i can tell, and that in fact the PLC responds with this error. I also checked the  Wireshark capture and also the Libnodave debug output, and could not find an error.

**Log files**
I captured this conversation with wireshark. It shows only the Upload packets, no connection packes:  
[Failure Upload pcap.zip](https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/files/505750/Failure.Upload.pcap.zip)

I also set the Debug output of libnodave to its maximum and activated an console to see if there is an error. I added some comments into the console output:
[Failure Upload.txt](https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/files/505752/Failure.Upload.txt)

jogibear9988, 2016-10-03:
I asked your bug someone at sps-forum who has more knowledge at the s7 protocol than I. seems i'm not using the upload id i got from the cpu. So it only works with the id 0. i'l look to fix it

jogibear9988, 2016-10-04:
can you try with the new dll's from here: https://ci.appveyor.com/project/jogibear9988/dotnetsiemensplctoolboxlibrary/build/artifacts

jogibear9988, 2016-10-07:
Does it work?

StefanHasensperling, 2016-10-11:
Sorry for the delay, I was traveling a lot lately. 
I don't have access anymore to the controller. But i think in the next two weeks i will be able to test with another controller of the same series.

StefanHasensperling, 2016-10-28:
I checked today and I can confirm that this bug is fixed.

But there is one thing that I want to change.
When the block upload fails, it should throw an exception instead if returning null.

I will propose this isnew an pull request when I have time.

StefanHasensperling, 2016-10-28:
This also fixed an error that I encountered just now, when multiple uploads where running at the same time. I tested with an previous version, and sometimes the upload would fail, when multiple uploads where running.
This seems to be fixed as well in the current version.

jogibear9988, 2016-10-28:
Thanks for your detailed report and tests. I will then close this issue!

closed 2016-10-28 by jogibear9988