dotnetprojects/DotNetSiemensPLCToolBoxLibrary #63: Start / Stop CPU

dgehri, 2016-10-20:
Hello - would it be possible to start & stop a S7-300 using this library?

StefanHasensperling, 2016-10-20:
Yes, this is possible using the functions:
public void PLCStart()
public void PLCStop()

But be advised that stopping and starting an controller may have consequences to the equipment controlled by the Controller.

dgehri, 2016-10-20:
Thank you very much! Could you elaborate on the consequences? Are there consequences different from starting / stopping the PLC using eg. Siemens TIA in online mode?

StefanHasensperling, 2016-10-20:
The functions are doing the same thing as TIA or Simatic manager does. 
The warning was meant as an warning for people that are not aware of what stopping an controller actually means, and try to play around on live machinery.

So in summary, these functions are doing exactly the same thing as TIA portal (or simatic manager).. There is also an function "PLCGetState"  which lets you query the current status of the controller (stopped, run, starting, etc.)

closed 2016-10-28 by jogibear9988