dotnetprojects/DotNetSiemensPLCToolBoxLibrary #66: Seperate Library from Windows.Forms

StefanHasensperling, 2016-11-02:
I think we should separate the library into two (or more) libraries. One with the library itself, one with all Windows Forms related stuff and maybe one for WPF as well. This way we could eliminate the dependency to Windows.Forms assembly in the main communications library.

Of course this is an breaking change and a lot of stuff need to be reworked to make this possible.
What do you think?

jogibear9988, 2016-11-07:
i will look! don't think it's so much work

jogibear9988, 2016-11-07:
Think it's a good idea, maybe i can refactor a little bit more! If i do, i will increase the main version, so we see the breaking change

jogibear9988, 2016-11-13:
The splitting was esier then thought, took only 15 minutes.

Now I have to do the nuget configuration right!

jogibear9988, 2016-11-13:
I've pushed it: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/tree/refactorV3

I will still do a bit more refactoring before releasing this!

jogibear9988, 2016-11-13:
I think, I also remove the compiled DLLs from the repo (this was before I used Appveyor & Nuget, so people could download a dll directly), and mybe split the examples out (but I'm not yet sure about that)

StefanHasensperling, 2016-11-13:
Just noticed an minor mistake of mine. The default com port speed for the MPI adapter. I think it is 19.2kbaud (at least if my memory is not failing me). but the default value is 38400.
So either the summary or the default value is not correct in the configuration class. 

 private string _comPortSpeed = "**38400**";
        /// <summary>
        /// defines the Bus speed for the connected MPI adapter
        /// The default MPI bus speed is **19200** bit / second
        /// </summary>
        public string ComPortSpeed

jogibear9988, 2016-11-19:
This has been done in V3 Branch.

jogibear9988, 2019-08-12:
will be fixed in #136

closed 2019-08-12 by jogibear9988