dotnetprojects/DotNetSiemensPLCToolBoxLibrary #71: Reading from a adjusted (Symbol or Comment) Instance DB

Timo1011, 2017-06-22:
Hello,

I found an issue while reading from an Instance DB.
I am using Step7 V5.5 + SP4.

After defining and calling a FB, Step7 and DotNetSiemensPLCToolBoxLibrary (ToolboxForSiemensPLCs) are showing the same content:
 
Step7:
[image]

ToolboxForSiemensPLCs:
[image]

After changing a Symbol or Comment in the FB, Step7 updates the corresponding DB:
[image]

When I open the DB with DotNetSiemensPLCToolBoxLibrary (ToolboxForSiemensPLCs), it returns still the old Symbol "InitVarname" and old Comment "Initial Comment". After a real change of the FB-interface (for example adding a parameter) and "Check and update Accesses", the Library shows the correct values. But, after a Symbol or a Comment change, an Access update will not be executed.

[S7_pro1.zip](https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/files/1094217/S7_pro1.zip)

StefanHasensperling, 2017-06-27:
I investigated this by reviewing the DBF file and i found the reason for this.
The Datenblocks information that is stored in the Project files is only actualized when an new Instance DB is created. Only then Step7 will create an new "block". When you change an comment in the FB afterwards, the generated instance DBs are not modified and still contain the comments that where current when they where generated. 

So, the Intance DBs code will only ever be actualized by Step7 when you create an new instance of an FB, either by call an FB with an new Instance DB number or when your change the Interface of the FB (then Step7 asks you to regenerate the Insance DBs of existing instances).

What the Step7 Code editor actually does, is it simply shows you the interface Declaration of the FB instead of the interface declaration stored for the instance DB. Meaning that the Code for an Instance DB and the FB may actually be different in the project files, Step7 just never shows you the DB's code. 

The fix is relatively simple. If the Block is an Instance DB, the use the declartion of the corresponding FB instead of the DB. I already implemented this in the Attached file in the function "GetBlockBytes". 

Hopefully later this week i will come around for creating an Pull Request, In the mean time you can just replaze the "BlocksOfflineFolder.cs" with the one I attached, or somebody else can make an Pull request out of this.

[BlocksOfflineFolder.zip](https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/files/1105988/BlocksOfflineFolder.zip)

jogibear9988, 2017-06-27:
If you create a pull req, can you include a switch, sonsomeone could also get the interface wich is in the current instance db? for example to restore some old data

jogibear9988, 2017-06-27:
and the actual data, from where do you read it? and what is if the fb was extended, does the read of the actual data from a smaler db work?

Timo1011, 2017-06-28:
Thank you very much. 
I will try it.

StefanHasensperling, 2017-07-18:
I tested it with the samples provided in the Issue that was raised. Everything is working fine. 
If the DB and FB are compatible, the FB interface is shown, if they are not compatible, the interface of the DB is shown. 

I dont know how to test the issue that you mentioned with the acutal data of the datablock. I think it should not be an problem, because Dataintegrity should always be ensured. If DB and FB are compatible, then so are the Actual data. If they are not compatible, then the DB is used as interface, for which the Actual data is compatible anyway.

Timo1011, 2017-07-18:
Hello Stefan,

 

I tested your changes with my S7 project and everything is working fine.

 

Thank you.

 

I have a question to another topic:

In parameter.cs the call of "FillActualValuesInDataBlock" is commented out. Because I need to read actual values, I activated this call and made some fixes on some other places in the Library. Do you know the history of this function and why is it commented out?

 

Do you or the forum have an interrest of my local changes?

 

Regards,

Timo

 

 

Gesendet: Dienstag, 18. Juli 2017 um 02:09 Uhr

closed 2017-07-21 by jogibear9988