dotnetprojects/DotNetSiemensPLCToolBoxLibrary #73: Reachable PLC discovery

StefanHasensperling, 2017-07-18:
I made an class that discovers available accesible CPU's (only by means of TCP/IP) on an network. This is currently part of an internal project of mine, but i want to integrate it into this Library. It comes with an WinForms dialog, that can be used.  It is tested with S7-300, S7-400 and IBH Softec Simulation PLC.

Problem is, it is written in VB.net, not C#. So there are two options. 
Should i Implement it as an sepárate Library, or Convert everything to C# to integrate it into this library (I dont know when y will have the time to do so). Which Option do you prefer, or do you have an better solution?

jogibear9988, 2017-07-18:
Maybe you can send it to me, I'll habe a look.
I'll try one of the Vb.net -> c# converters

jogibear9988, 2017-07-18:
or compile it, and use ilspy or dotpeek to extract e csharp project

StefanHasensperling, 2017-07-21:
I adjusted the Code a little bit, and then translated it via Telerik online converter. The Output was quite good, and i only needed to change small errors. The Pull request is ready. Also i fixed an bug in MC7 to AWL parsing of Interface Declarations.

closed 2017-07-21 by jogibear9988