dotnetprojects/DotNetSiemensPLCToolBoxLibrary #83: Bug in Block Interface Parsing from MC7

StefanHasensperling, 2017-09-21:
Today y found some bugs in the Interface Parsing section of the MC7Converter, namely the in Parameters.cs. When y did some testing with some unusual code code blocks from one of my clients. 
I already Identified them, but more testing needs to be done. 

I found the following Bugs:
-When BLOCK_SFB are used in the Interface, for example when calling SFB14 "GET" as MuiliInstances, then the code enters an Endless loop. 
-The handling of BLOCK_xx Interface declarations is not correct. So if one of these is used, it Parsing will probably fail. 

I am already putting together an Pull Request, for fixing the issues, especially the Endless loop thing.

StefanHasensperling, 2017-09-21:
Both should be fixed

closed 2017-09-21 by StefanHasensperling

jogibear9988, 2017-09-21:
What do you use to Program? Visual Studio? Do you need a Resharper Licencence for your Open Source work?

StefanHasensperling, 2017-09-21:
I am using VS Community 2017, with just a few extension. 
An license would be great!

jogibear9988, 2017-09-21:
can you send me a email?