dotnetprojects/DotNetSiemensPLCToolBoxLibrary #91: Bausteinattribute von S7Classic

stollenkroll, 2018-02-13:
Hallo,
ich möchte von S7-Bauteinen die Attribute (Erstellsprache, Version, Name (Header), Autor, Familie, Änderungsdatum, letzte Schnittstellenänderung) auswerten.
Leider bekomme ich keine oder falsche Werte angezeigt.

Meine Coderealisierung:
`For Each baustein In s7ProgrammOrdner.BlocksFolder.BlockInfos`
`If baustein.blockname = "FC4" Then`
                        `block= baustein.GetBlock`
                    `End If`

Attributergebnisse:

SimaticManager  | PLCToolBoxLibrary | Ergebnis | Objekt
---------------- | ------------------- | --------- | --------
Erstellsprache | BlockLanguage | Nothing | DataTypes.Blocks.Block).BlockLanguage
Version | Version | 12.11 Falscher Wert (11.15) | DataTypes.Blocks.Step7V5.S7Block).Version
Name | Name | Nothing | DataTypes.Blocks.Block).Name
Autor | Autor | IO
Familie | Family | Nothing | DataTypes.Blocks.Step7V5.S7Block).Family
Änderungsdatum | LastCodeChange | Bei UDT's #1/1/0001 12:00:00 AM# | DataTypes.Blocks.Step7V5.S7Block).LastCodeChange
letzte Schnittstellenänderung | LastInterfaceChange | Bei UDT's #1/1/0001 12:00:00 AM# | DataTypes.Blocks.Step7V5.S7Block).LastInterfaceChange

 Hier die Quelle zum BeispielFC4:

>FUNCTION "Symbolischer Name" : VOID
>TITLE =Bausteintitel
>//BausteinKommentar
>//Zeile2
>AUTHOR : Autor
>FAMILY : Familie
>NAME : 'Name'
>VERSION : 11.15
>
>BEGIN
>NETWORK
>TITLE =Netzwerkkommentar
>
>      U     "E_1.0"; 
>      =     "m_cam_active"; 
>END_FUNCTION

Was mache ich falsch?

StefanHasensperling, 2018-02-16:
Hallo, Sorry fuer die Verspätung. Bin Grad Im Urlaub.
Wenn du das Projekt mit der Bibliothek geöffnet hast, sollte das eigentlich
so funktionieren.
Ich werde leider erst Ende nächster Woche dazu kommen mir das anzuschauen.

jogibear9988, 2018-03-15:
Gibts hierzu Neuigkeiten? Hat das funktioniert?

stollenkroll, 2018-03-16:
Leider nein. Da ich noch keine weiteren Informationen dazu habe, wie ich die Attribute erhalte, habe ich ersteinmal mein S7-Programm mit der Simatic-Lib analysiert.

jogibear9988, 2020-11-07:
sollte gefixt sein

closed 2020-11-07 by jogibear9988