dotnetprojects/DotNetSiemensPLCToolBoxLibrary #95: tia portal file size changes

impoetk, 2018-03-20:
In the directory i see this file with 571 kb, in a vm with win7 x64 fresh installation, i also see 571 kb as size

DotNetSiemensPLCToolBoxLibrary/externalDlls/siemens/V15/Siemens.Engineering.dll

but in recent installations for V15 (win 7 and win 10 both x64) i've seen files of size 586 kb, and for this files i have an exception (type initialization exception) for type Siemens.Engineering.TiaPortal

both have file versions equal (from the properties of the file) all installations have the same product versions under details of installed software (tia portal dialog)

Should the dll be updated with the one of 586 kb???

jogibear9988, 2018-03-20:
Maybe File changes in Update1 for V15?

I thought the 586 file throws an exception?

impoetk, 2018-03-21:
yes it throws an exception... but i dont know why

impoetk, 2018-03-21:
is it possible to parse the binary file for the project? i saw some code that started doing this but is commented, it would really help if its available at least for variables, maybe you can give me some hints and i can continue working on it

jogibear9988, 2018-03-21:
I tried, but had no luck... And atm. I've no time to look for it!

What I've discoverd, I put in a plf grammer wich hexinator can use: https://www.sps-forum.de/hochsprachen-opc/76451-hex-editor-mit-gramatik-zur-fileanalyse-tia-file-format-grammar-file.html#post533281

impoetk, 2018-03-22:
i started reverse engineering the plf file... im amazed from what i found so far
a - started with a file with a plc of address 10.10.10.10, save the file in other location
b - changed ipaddress to 10.10.10.11, save the file in other location
c - changed ipaddress back to 10.10.10.10 save the file in other location

make hex dif of files
a -> b 765 bytes added at the end
b -> c 783 bytes added at the end
a -> c 1548 bytes added at the end

impoetk, 2018-03-22:
i've never seen the openess scripter from siemens... do you think is possible to wrap this in your library???

jogibear9988, 2018-03-22:
as I said... if you try to analyze the tia file format, try to use my grammer as a start. and use code in my lib to  create xml file from the project file, and try to check out what in the xml is where in the file

jogibear9988, 2018-03-22:
why should i include the scripter in my lib? what would be the advantage?

impoetk, 2018-03-22:
the scripter is able to load different instances of tia portal without restarting, has already code to initialize the portal instance with a enum parameter of the version you wish to do, checking if the software is installed, if the current user is already part of tia openness groups, and the list increases....

jogibear9988, 2018-03-22:
i don‘t think we are allowed too use the scripter dlls.

Are you sure that the scripter does not run a extra process/appdomain while running the script?

impoetk, 2018-03-27:
sorry, i dont know for sure, when it is installed it the folder have 2 .exe, so probably is using separate process

closed 2018-03-27 by impoetk