dotnetprojects/DotNetSiemensPLCToolBoxLibrary #99: S7-1500 support

abrasat, 2018-05-03:
Does this library support also S7-1500 ?

jogibear9988, 2018-05-03:
What do you mean? S7-1500 Communication? If so, yes, but only the old S7 Protocol, means Absolute Addressed Variables.
If you mean TIA File access, then yes, but only when tia is installed. It acts as an wrapper to openness, so you can use the same api for Step7 Projects and TIA Projects

Direct File Access was started, but the TIA File Format is so complex and closed, it's a too big taks to do it as a side project!

Communication to S7 1500 with the new protocol was never started. But if there are Infos how to do, I will implement it!

abrasat, 2018-05-04:
Thank you for the answer.
Yes, communication with S7-1500.
The application should run also on PCs without TIA installation.

closed 2018-05-04 by jogibear9988