dotnetprojects/DotNetSiemensPLCToolBoxLibrary #105: Socket Leak in PLCConnection.Connect()

ledbette, 2018-11-28:
The socket connection code will leak sockets if unable to connect to the PLC within the specified timeout:

                    case LibNodaveConnectionTypes.ISO_over_TCP:
                    case LibNodaveConnectionTypes.ISO_over_TCP_CP_243:
                    case LibNodaveConnectionTypes.Netlink_lite:
                    case LibNodaveConnectionTypes.Netlink_lite_PPI:
                    case LibNodaveConnectionTypes.Netlink_Pro:
                        _errorCodeConverter = libnodave.daveStrerror;
                        socketTimer = new System.Timers.Timer(_configuration.TimeoutIPConnect.TotalMilliseconds);
                        socketTimer.AutoReset = true;
                        socketTimer.Elapsed += socketTimer_Elapsed;
                        socketTimer.Start();
                        socketThread = new Thread(this.socket_Thread);
                        socketThread.Start();
                        try
                        {
                            while (socketThread.ThreadState != ThreadState.AbortRequested &&
                                   socketThread.ThreadState != ThreadState.Aborted &&
                                   socketThread.ThreadState != ThreadState.Stopped)
                            {
                                Thread.Sleep(50);
                            }
                        }
                        catch (Exception ex)
                        {
                        }
                        if (socketTimer != null)
                            socketTimer.Stop();
                        if (socketThread != null)
                            socketThread.Abort();
                        socketTimer = null;
                        socketThread = null;

In this code, if a connection timeout occurs, the connection thread is aborted, but the libnodave.openSocket() call appears to continue trying to connect until the SYN timeout occurs.  If the PLC becomes available within the SYN timeout period (typically 20 seconds), libnodave will connect, but the calling thread has moved on and the resulting socket connection is never closed. 

If the calling application attempts to reconnect multiple times until the connection is successful, you will see a number of socket connections get leaked. This is evident by using netstat and/or TCPView and viewing the number of established connections held by the client application.

To remedy this problem, I created an internal class LibNoDaveConnectionHelper.  In this class, the socket connection is attempted using a task, and I eliminated the separate timer thread by using a Task.Wait() call.  If the wait times out, I set an abort flag, but allow the connection task run to completion.  If the connection completes successfully, but the application has aborted, the socket is closed.

To use it, replace the above socket connection code by the following.

                        _fds.rfd = LibNoDaveSocketConnectHelper.Connect(_configuration);

The connection helper class is as follows:

    /// <summary>
    /// LibNoDave Connection Helper
    /// </summary>
    internal class LibNoDaveSocketConnectHelper
    {
        public static readonly IntPtr TimeoutSocketPtr = new IntPtr(-999);
        private IntPtr m_socketPtr = TimeoutSocketPtr;
        private bool m_isAborted;
        private readonly object m_syncLock = new object();

        /// <summary>
        /// Connect Socket
        /// </summary>
        /// <param name="configuration">PLC Connection Configuration</param>
        /// <returns>Socket Pointer or IntPtr(-999) if connection timed out</returns>
        public static IntPtr Connect(PLCConnectionConfiguration configuration)
        {
            var helper = new LibNoDaveSocketConnectHelper();

            // perform a name lookup if necessary
            string hostAddr = configuration.CpuIP;
            try
            {
                IPAddress[] addresslist = Dns.GetHostAddresses(hostAddr);
                hostAddr = addresslist.First(ipAddress => ipAddress.AddressFamily == AddressFamily.InterNetwork).ToString();
            }
            catch (Exception ex)
            { }

            return helper.DoConnect(hostAddr, configuration.Port, configuration.TimeoutIPConnect.Milliseconds);
        }

        /// <summary>
        /// Private Constructor
        /// </summary>
        private LibNoDaveSocketConnectHelper()
        {
        }

        /// <summary>
        /// Run the connection task in the background
        /// </summary>
        /// <param name="port">TCP Port</param>
        /// <param name="connectTimeoutMsec">Connection Timeout in Milliseconds</param>
        /// <param name="ipAddr">Host IP Address</param>
        /// <returns>Socket Pointer or IntPtr(-999) if connection timed out</returns>
        private IntPtr DoConnect(string ipAddr, int port, int connectTimeoutMsec)
        {
            // start connection thread
            Task task = Task.Run(() => ConnectTask(ipAddr, port));

            if (!task.Wait(connectTimeoutMsec))
            {
                lock (m_syncLock)
                {
                    m_isAborted = true;
                    return TimeoutSocketPtr;
                }
            }

            return m_socketPtr;
        }

        /// <summary>
        /// Connection Task
        /// </summary>
        private void ConnectTask(string hostAddr, int port)
        {
            IntPtr socketPtr = libnodave.openSocket(port, hostAddr);

            lock (m_syncLock)
            {
                if (m_isAborted)
                {
                    // if we've aborted, wait until the connection attempt completes and close the socket connection otherwise we'll leak a socket connection
                    libnodave.closeSocket(socketPtr);
                }
                else
                {
                    m_socketPtr = socketPtr;
                }
            }
        }
    }

ledbette, 2018-12-11:
I'm also seeing some additional socket leaks when I have a noisy or bad ethernet connection.  It appears these are down in the libnodave library, but I've yet to isolate them.  Any additional insight you might be able to provide would be appreciated.

jogibear9988, 2018-12-11:
I think we could also get the socket with managed code via tcpclient. maybe this will work better

ledbette, 2018-12-11:
We have a client that is working in a noisy network environment, which causes loss of TCP connections. These lost connections occasionally leak sockets.  It's not every time and I haven't fully characterized it.
I found one place up in the PLCToolbox library that was causing a leak on connection - the comment I posted earlier. But that didn't completely fix the problem.

What seems to happen is if the connection drops early (within the first 5 packet transmission), it seems that libnodave leaks a socket.  I don't see the connect call being successful on the PLCToolbox side of the library, but it appears a socket connection is made in libnodave, but leaked immediately.  So, I'm guessing some sort of error occurs in the connection, but the socket connection was at least partially established such that the socket is allocated by Windows and Windows continues to hold the connection to the PLC with keepalives. (I haven't snooped the ethernet traffic, I'm just guessing).
I have a co-worker that made an arduino-triggered relay we use to simulate the problem.  The relay breaks the ethernet connection at a specified interval.  If the interval is set to 3 seconds on/3 seconds off, the library leaks sockets quite frequently (about one every minute or so). Our PLC is configured to allow up to 16 connections, so once that 16th socket is leaked, you lose comm to the PLC.

I'm using ISO_over_TCP (connection type 122).  I tried to switch to using the ISO_over_TCP_Managed, but was unable to get a connection to the PLC at all.
I'm a bit at a loss as to where these socket leaks are occurring.  Since it doesn't appear libnodave is being actively supported any longer, I'm starting to dig into the libnodave source code today.
Wade

jogibear9988, 2018-12-11:
I don't think it's a libnodave problem. 

The closeSocket is never called inside of nodave.c, so it's completly up top the library user to close the socket.
I think it's an error in my connection wrapper. Socket is only closed on dispose of connection

jogibear9988, 2018-12-11:
Can you check if my fix works for you?

ledbette, 2018-12-11:
It will be Thursday before I can get back into the test environment where I can duplicate the problem.  I have also found another location in my code where I think it could potentially be leaking.  I'll let you know what I find out.  
Thanks for your responsiveness.
Wade

jogibear9988, 2019-01-01:
thx, hopefully they are solved

closed 2019-02-07 by jogibear9988

ledbette, 2019-02-22:
Sorry, I was never able to get back onto this and test it.  It's not something I work on actively, so don't have an environment to test in.  I have to rely on one of my clients.
I do have a question that maybe you can help with.  
We're using ISOTCP (122).
We've had some cases where we have lost connection to the PLC, and are unable to re-establish.  We get a TCP socket connection, but the ISO-level connection seems to be failing.  _dc.connectPLC() returns -1. 
The PLC is still running, as other systems are able to communicate with it.  If we kill the application (my application is an OPC Server), we are able to re-establish comm with the PLC.  So, that would seem to indicate it's something on our side.  I'm thinking something isn't getting cleaned up.
Any ideas of what to look at?  
I don't really any experience with the ISO-level comm with the S7 (but it appears I might be getting some soon).

 
 
Closed #105.

jogibear9988, 2019-02-22:
maybe it‘s better to raise a fresh issue.
this seems to be relatedto this issue! do you have this problem with your modifications, or with the current version of the library?

ledbette, 2019-02-22:
I agree, it's not directly related to this issue. I'm just looking for some troubleshooting tips to start with. I'm using a modified version of the library with enhanced logging, and I have an OPC Server application around it controlling connects, disconnects, writes and read polling.I'm trying to identify a reproducible scenario in which the problem occurs. All I have at this point is the users come in after running overnight and the client can't connect. We have multiple PCs communicating with the PLC simultaneously. A couple will be unable to connect, while a third will be working fine.I've added logging to try to understand the conditions. Once I identify a reproducible scenario. I'll open a new ticket to try to address the cause.Wade

jogibear9988, 2019-02-22:
and your modifications include my fixes I asked you to test?

ledbette, 2019-02-22:
Not yet.  I'm running the code I had submitted in the issue.
I will be at my client this afternoon (I'm in Dallas, TX, USA), and should be able to pull in your fixes to test.
Wade