dotnetprojects/DotNetSiemensPLCToolBoxLibrary #107: Frage NCK Daten Lesen / Schreiben (840dsl)

KES-CK, 2019-02-11:
Ich ich habe der DotNetSiemensPLCToolBoxLibrary Beispiele gefunden um NCK Variablen zu lesen.
Ich habe eine 840Dsl Steuerung und versuche R-Parameter zu lesen.

Beim Lese-Zugriff erhalte ich folgende Exception:
Operation failed due to error from PLC 33028: context is not supported. Step7 says:Function not implemented or error in telgram.

Was mache ich falsch, bzw, gibt es noch weitere Randbedingungen?

Beispielcode:

                //data from nc var selector
                //STRUCT
                // SYNTAX_ID : BYTE := B#16#82;
                // bereich_u_einheit : BYTE := B#16#41;
                // spalte : WORD := W#16#1;
                // zeile : WORD := W#16#1;
                // bausteintyp : BYTE := B#16#15;
                // ZEILENANZAHL : BYTE := B#16#1;
                // typ : BYTE := B#16#F;
                // laenge : BYTE := B#16#8;
                //END_STRUCT ;
            try
            {
               NC_Var rParam1_1 = new NC_Var(0x82, 0x41, 0x1, 0x1, 0x15, 0x1, 0xF, 0x8);
                myConn.ReadValue(rParam1_1);
            }
            catch (Exception ex)
            {
                MessageBox.Show(ex.ToString());
            }

Mfg

Christian Kunz

jogibear9988, 2019-02-11:
Kann ich nicht wirklich weiterhelfen.
Schau und frag mal hier: https://www.sps-forum.de/hochsprachen-opc/80971-dotnetsiemensplctoolboxlibrary-libnodave-zugriff-auf-dual-port-ram-fb15.html

Nick135, 2019-02-11:
Wie hast du "myConn" konfiguriert?

KES-CK, 2019-02-11:
Hallo Zusammen,

danke für die schnellen Antworten. Den Thread im SPS Forum werde ich jetzt mal durcharbeiten. Meine Verbindung habe ich folgendermaßen konfiguriert:

Die 192.168.214.1 ist die IP-Adresse der 840Dsl der Zugriff auf die PLC klappt ohne Probleme.
[image]

Nick135, 2019-02-11:
Nimm mal Slot 4, dann wird es funktionieren

KES-CK, 2019-02-11:
Hey Super,

werde ich direkt testen wenn ich wieder im Büro bin.
Klappt über den Slot 4 auch noch der Zugriff auf die PLC oder muss ich dann wieder über
Slot 2 zugreifen?

Gruß Christian

Nick135, 2019-02-11:
Für die PLC brauchst du Slot 2 und für die Nck Slot 4.

Ich bau die Verbindung ohne die grafische Eingabe auf.

`using (var myConn = new DotNetSiemensPLCToolBoxLibrary.Communication.PLCConnection("test"))
{
      myConn.Configuration.CpuIP = !string.IsNullOrEmpty(IPAddress) ? IPAddress : "192.168.214.1";
      myConn.Configuration.CpuSlot = plcConnection ? 2 :  4;
      myConn.Connect();
      var r1 = myConn.ReadValue(new NC_Var(0x82, 0x41, 0x1, 0x1, 0x15, 0x1, 0xF, 0x8));
}`

KES-CK, 2019-02-11:
Alles klar, danke nochmal für die schnelle Hilfe!

jogibear9988, 2019-02-11:
@KES-CK du kannst natürlich mehrere Connections gleichzeitig anlegen!

@Nick135 soll ich dich als maintainer hier hinzufügen?

KES-CK, 2019-02-12:
@jogibear9988 Ich werde das Testen und eine Rückmeldung geben ob alles geklappt hat.

Nick135, 2019-02-12:
@jogibear9988 bei allen Steuerungen die ich je getestet habe war es so, Bei AG-Link wird, wenn ich mich richtig erinnere, der Slot 3 verwendet.

Slot kann der Hardwarekonfig entnommen werden.
[image]

KES-CK, 2019-02-19:
Hallo zusammen,
entschuldigt die späte Rückmeldung. Aber bin heute erst wieder an die Test-Steuerung gekommen.
Die Verbindung sowie Lese- und Schreibvorgänge klappen über Slot 4.

Jetzt teste ich noch Zugriffe / weitere Funktionen.
Kann mir jemand sagen wie ich ein PI-Dienst gestartet bekomme.
z.B. LOGIN oder LOGOUT
Ich habe schon diverse Tests gemacht, die aber immer in einer Exception geendet sind.
hier ein paar meiner Versuche:
`_myConn.PI_Service("_N_LOGIN_", new string[] { "P01","SUNRISE" });`
`_myConn.PI_Service("_N_LOGIN_", new string[] { "SUNRISE" });`
`_myConn.PI_Service("_N_LOGOFF", new string[] { "" });`

closed 2019-08-12 by jogibear9988