dotnetprojects/DotNetSiemensPLCToolBoxLibrary #135: S5 Siemens PLC - Reading Outputs Doesn't Work

InnovationUnplugged, 2019-08-09:
The issue that i have is that i can't read digital outputs on a S5. After calling ReadValues("A32.0") i always get false even though the status leds on the PLC are changing from False -> True. Is that a known issue? Is there a code snippet that solves that problem?

jogibear9988, 2019-08-09:
I‘ve never tested on a S5 (I don‘t have one). I only implemnted it cause it was supported in libnodave

jogibear9988, 2019-08-11:
@InnovationUnplugged how do you connect your S5?

jogibear9988, 2019-08-16:
no answers

closed 2019-08-16 by jogibear9988

InnovationUnplugged, 2019-08-16:
Then how about this: Since i have the S5 PLC and i have developers background - i can contribute by testing the code on my PLC. Currently i connect to the PLC using the PG port and an active tty > RS232 converter. As far a the code goes - i can start and stop the PLC, but when i try to read out any of the digital Outs i always get a False. So what do you say ? Do we try to tackle this issue?

InnovationUnplugged, 2019-08-16:
i can also share code snippets of what i am trying to do.... i have to say that enabling this Digita output readouts will be a great added value to this project from my point of view.... but i need somebody on your side to support me with this.... I can offer quite a lot of my time + PLC + testing approach.

jogibear9988, 2019-09-10:
@InnovationUnplugged sorry I haven't seen your answers here :-(