dotnetprojects/DotNetSiemensPLCToolBoxLibrary #139: Build Error

Nick135, 2019-09-06:
Weder mit VS2019 (16.2.4) noch VS2019 Preview (16.3.0 Preview 3.0) kann das Projekt erstellt werden.

> Error	CS0006	Metadata file '...\dotnetprojects\DotNetSiemensPLCToolBoxLibrary\LibNoDaveConnectionLibrary\bin\Debug\netstandard2.1\DotNetSiemensPLCToolBoxLibrary.dll' could not be found	TiaImporter (netcoreapp3.0)	...\dotnetprojects\DotNetSiemensPLCToolBoxLibrary\TiaImporter\CSC	1		Active

> Error	MSB3041	Unable to create a manifest resource name for "DataTypes\DiagnosticEntry.resx". Could not find a part of the path '...\dotnetprojects\DotNetSiemensPLCToolBoxLibrary\LibNoDaveConnectionLibrary\DataTypes\DataTypes\DiagnosticEntry.cs'.	DotNetSiemensPLCToolBoxLibrary	C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\MSBuild\Current\Bin\Microsoft.CSharp.CurrentVersion.targets	100		

> Error	MSB4018	The "GenerateResource" task failed unexpectedly.
System.NullReferenceException: Object reference not set to an instance of an object.
   at Microsoft.Build.Tasks.ProcessResourceFiles.get_HaveSystemResourcesExtensionsReference()
   at Microsoft.Build.Tasks.ProcessResourceFiles.WriteBinaryResources(ReaderInfo reader, String filename)
   at Microsoft.Build.Tasks.ProcessResourceFiles.WriteResources(ReaderInfo reader, String filename)
   at Microsoft.Build.Tasks.ProcessResourceFiles.ProcessFile(String inFile, String outFileOrDir)
   at Microsoft.Build.Tasks.ProcessResourceFiles.Run(TaskLoggingHelper log, ITaskItem[] assemblyFilesList, List`1 inputs, List`1 satelliteInputs, List`1 outputs, Boolean sourcePath, Boolean usePreserializedResources, String language, String namespacename, String resourcesNamespace, String filename, String classname, Boolean publicClass, Boolean extractingResWFiles, String resWOutputDirectory)
   at Microsoft.Build.Tasks.GenerateResource.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	DotNetSiemensPLCToolBoxLibrary.WinForms	C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\MSBuild\Current\Bin\Microsoft.Common.CurrentVersion.targets	3056

jogibear9988, 2019-09-06:
netcore3 preview7 muss installiert sein!

Nick135, 2019-09-06:
Hab netcore3 preview9 installiert

jogibear9988, 2019-09-06:
d.h es geht oder nicht? editier mal die global.json

Nick135, 2019-09-06:
d.h die Fehlermeldung ist mit installiertem netcore 3 preview9 uns vs2019 preview 3.0 gekommen.
Mit normalen vs2019 gab's noch mehr Fehler. 
Projekt ist direkt so wie es ausgecheckt wurde.

Was muss in der global.json angepasst werden?

jogibear9988, 2019-09-06:
die preview version, ist im moment 7

Nick135, 2019-09-06:
Du meinst aktuelle preview deinstallieren und die 7 installieren hilft?

jogibear9988, 2019-09-06:
nein, änder einfach in der global.json die 7 auf eine 9
https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/blob/master/global.json

jogibear9988, 2019-09-06:
@Nick135 does it work?
Will you create a pull req with your changes?

Nick1357, 2019-09-09:
Tut trotzdem nicht.

Wenn ich nur die LibNoDaveConnectionLibrary compiliere kommt dieser Fehler:

> Error	MSB3041	Unable to create a manifest resource name for "DataTypes\DiagnosticEntry.resx". Could not find a part of the path 'C:\...\DotNetSiemensPLCToolBoxLibrary\LibNoDaveConnectionLibrary\DataTypes\DataTypes\DiagnosticEntry.cs'.	DotNetSiemensPLCToolBoxLibrary	C:\Program Files (x86)\Microsoft Visual Studio\2019\Preview\MSBuild\Current\Bin\Microsoft.CSharp.CurrentVersion.targets	100

Woher kommt die Pfadangabe "LibNoDaveConnectionLibrary\DataTypes\DataTypes\DiagnosticEntry.cs"? Datei liegt in 
LibNoDaveConnectionLibrary\DataTypes\DiagnosticEntry.cs

Nick135, 2019-09-09:
Tut trotzdem nicht.

Wenn ich nur die LibNoDaveConnectionLibrary compiliere kommt dieser Fehler:

> Error MSB3041 Unable to create a manifest resource name for "DataTypes\DiagnosticEntry.resx". Could not find a part of the path [...]

Woher kommt die Pfadangabe "LibNoDaveConnectionLibrary\DataTypes\DataTypes\DiagnosticEntry.cs"? Datei liegt in
LibNoDaveConnectionLibrary\DataTypes\DiagnosticEntry.cs

jogibear9988, 2019-09-09:
kp... auf appveyor läufts ja auch: https://ci.appveyor.com/project/jogibear9988/dotnetsiemensplctoolboxlibrary/builds/27139407

jogibear9988, 2019-09-09:
hab nun auch preview 9 installiert... funktioniert

Nick135, 2019-09-12:
Hab ein wenig im Internet gesucht und die Lösung ist das erweitern der PropertyGroup der DotNetSiemensPLCToolBoxLibrary.csproj

`<EmbeddedResourceUseDependentUponConvention>false</EmbeddedResourceUseDependentUponConvention>`

So funktionierts:
`  <PropertyGroup>
    <GenerateErrorForMissingTargetingPacks>false</GenerateErrorForMissingTargetingPacks>
    <EmbeddedResourceUseDependentUponConvention>false</EmbeddedResourceUseDependentUponConvention>
  </PropertyGroup>`

Nick135, 2019-09-12:
Ich werd das mal testen und beobachten, jedoch mit meiner Version auf DotNet 4.0 bleiben. Das Funktioniert bis jetzt immer (auch bein WinXP Anlagen die auch noch die nächsten Jahre laufen werden)

closed 2020-04-24 by jogibear9988