dotnetprojects/DotNetSiemensPLCToolBoxLibrary #140: DotNetSiemensPLCToolBoxLibrary with S5 Siemens 95U using AS511

InnovationUnplugged, 2019-09-10:
Dear,
I am having issues for quite a long time already trying to read out digital outputs from a S5 Siemens PLC using AS511 protocol.
I created a new console application to present to you clean code.
Steps that i took:
1) Create a new .NET C# console application
2) Installed DotNetSiementPLCTookBoxLibrary from Nuget (version 4.0.347)
3) Created a small program that connects to the PLC and tries to read out A32.1 (DIGITAL OUTPUT 2)
4) Here is the code:

 static void Main(string[] args)
        {
            var config = new PLCConnectionConfiguration() {
                ComPort = "COM4",
                ComPortParity = LibNodaveConnectionBusParity.even,
                ComPortSpeed = "9600",
                ConnectionName = "AS511",
                ConnectionType = LibNodaveConnectionTypes.AS_511
            };

            var connection = new PLCConnection(config);
            
            connection.Connect();
            var tag = new PLCTag("A32.1");
            connection.ReadValue(tag);
            

            Console.ReadLine();
        }

Issue:
In all cases, no matter what i try to do, the program always read out the value of that digital output as FALSE, while i see it changing on the PLC from False to True and from True to False (the PLC is configured to change those outputs within randomized time frames).

Can somebody please point me towards the right direction with this?
Has anybody have some code snippets that would read out the OUTPUTS from that PLC?

Any suggestions are welcome :)

jogibear9988, 2019-09-10:
maybe it's this issue:

https://www.sps-forum.de/hochsprachen-opc/12187-libnodave-und-as511.html
https://www.sps-forum.de/simatic/12236-libnodave-s5.html

jogibear9988, 2019-09-10:
I tried to create a fix here: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/pull/141

jogibear9988, 2019-09-10:
please test with the libnodave_jfkmod from this artifacts https://ci.appveyor.com/project/jogibear9988/dotnetsiemensplctoolboxlibrary/builds/27313511

InnovationUnplugged, 2019-09-10:
HI Jogibear9988,
I will dig into your findings. so far i think that this is a fix for the underlying libnodave library written in C correct?
So far, when i connect to the PLC i get this information:

`connect::
                            0:0x00,0x00,0x00,0x00,0x00,0x63,0x00,0x63,0x80,0x61,0x00,0x5F,0x00,0x60,0x00,0x5D,
                            10:0x00,0x28,0x01,0x40,0xFF,0x00,0x00,0x02,0x00,0x02,0x00,0x02,0x00,0x02,0x00,0x02,
                            20:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0xFF,0x01,0x00,0x0A,0x00,0x00,0x10,0x03,
start of inputs in memory 6300
start of outputs in memory 6380
start of flags in memory 6100
start of timers in memory 5f00
start of counters in memory 6000
start of system data in memory 5d00`

which i think indicates that the initial handshake and the PLC -> PC communication is working...

jogibear9988, 2019-09-10:
yes, pleas test with the dll when the build is done on appveyor

InnovationUnplugged, 2019-09-10:
i went to the url you gave me and thre are artifacts left since they are removed after 6 months...

 Artifacts older than 6 months are automatically deleted. For more details see Artifacts retention policy.

jogibear9988, 2019-09-10:
?? the build is not yet finished

jogibear9988, 2019-09-10:
now it's done
https://ci.appveyor.com/project/jogibear9988/dotnetsiemensplctoolboxlibrary/builds/27313511/artifacts

InnovationUnplugged, 2019-09-10:
grande kudos! everything works like a charm. i would like to thank you for your interest and input - otherwise it would have been one uphill battle for me :)
One last question - will that fix be implemented in the main branch as well or do i have to store those 2 dlls (32 and 64 bit) that you have created in a safe so that i have them in the future?

jogibear9988, 2019-09-10:
i'll merge now

jogibear9988, 2019-09-10:
nuget should be automatically updated within 15 miniutes or so

closed 2019-09-10 by jogibear9988