dotnetprojects/DotNetSiemensPLCToolBoxLibrary #145: Read multiple NC_Var variables at once

bsantamariatekniker, 2020-02-06:
Hello everyone, I'm working on a project where I have to get variables from a 840d whose values we need to refresh as fast as possible.

The variables are the following ones:

- For each axis (6):
  - Set-point position
  - Real position measured on first encoder
  - Real position measured on second encoder
  - Velocity
  - Torque
- NC current program name

We have achieved this by using *Synchronized Actions* to get the data from the NC to the PLC and then retrieving them as an array using this library. We are able to refresh the variables each 50ms more or less, which is enough for our purposes.

We are now trying to repeat this but without modifying the PLC, by using NC_Vars from this library, but we are only able to refresh the variables at a 300-400ms period.
I think that one of the biggest limitations here is that we have to make a call to *ReadValue* for each variable, so we have to make 30 calls to the function.

```C#
for(var ii = 0; ii < _ncVarAddress.Count; ii++)
{
    values[ii] = (double)_nc.ReadValue(_ncVarAddress[ii]);
}
```

Is it possible to get more than one NC_Var variable for each *ReadValue* call? If not, do you know of any other way to get that variables at high frequency without using *Synchronized actions*?

Thanks in advance

jogibear9988, 2020-04-24:
I could not help with this.  Maybe @Nick135  could help

Nick135, 2020-04-30:
If you use PlcNck Tag in a List and then use connection.ReadValues(yourList) you get all data within one package
Limitations are the same as if you read/write Nck Variables from PLC.
- Only one channel is allowed
- Only one Axis area is allowed
- Area General and channel 1 can read together
- Only one Drive area is allowed

closed 2020-10-20 by jogibear9988