dotnetprojects/DotNetSiemensPLCToolBoxLibrary #147: Event Handler ValueChanged not working for S5 AS511

InnovationUnplugged, 2020-02-24:
Dear Community
I am working on a program to read out Digital Inputs and Outputs from S5 using AS511 protocol.
I created a C# program that goes like this:

PLCConnection _connection;
var config = new PLCConnectionConfiguration()
                {
                    ComPort = device.COMPort,
                    ComPortParity = device.ComPortParity == "even"
                        ? LibNodaveConnectionBusParity.even
                        : LibNodaveConnectionBusParity.none,
                    ComPortSpeed = device.ComPortSpeed.ToString(),
                    ConnectionName = device.DeviceName,
                    ConnectionType = device.ConnectionType == "AS_511"
                        ? LibNodaveConnectionTypes.AS_511
                        : LibNodaveConnectionTypes.ISO_over_TCP

                };
                _connection = new PLCConnection(config);

                _connection.Connect();

var tagList = new List<PLCTag>();
tagList.Add(new PLCTag("A32.1"));
tagList.Add(new PLCTag("E32.2"));

foreach(var tag in tagList)
{
tag.ValueChanged += (sender, e) => PlcTag_ValueChanged2Console(sender, "S5");
}

while (true)
{_connection.ReadValues(tagList);}

... and the EventHandler

private static void Tag_ValueChanged2Console(PLCTag sender, string deviceName)
{
            Console.WriteLine($@"{DateTime.Now:HH:mm:ss.ffffff} - Console - {deviceName} - {sender.S7FormatAddress} - {sender.Value}");
}

The digitalinput A32.1 is wired to digitalOutput E32.2 so when one changes, the other one react immediately.

What i noticed:
Event Handler (above) is being raised randomly. It should fire twice for 1 change from true to false of the output since that output is connected to an input.

Question:
Why is this happening?
Since i have to read status of all DI and DO upon every change of a single DI or DO - am i using the best method for that? What is the fastest way or reading the status of multiple DIs / DOs.

Regards

InnovationUnplugged, 2020-02-24:
I should also add that the amount of DO and DI that i need to read goes in hundreds. 
For development purposes i connected the event handler to just 2 signals (which are wired together) but the event handler is firing for only the first signal defined, the second - wired signal is not raising the event handler.
I can gladly share my whole source code of the application in order to provide better insights to what i am doing.

@jogibear9988 - last time you were very helpful with my previous issue - do you have any golden thoughts about this issue?

Regards

jogibear9988, 2020-04-24:
Sorry if not seen this.

Is this still a problem?

jogibear9988, 2020-04-24:
But I've no Idea whats wrong.
I also could not test AS511

jogibear9988, 2020-11-04:
no answer

closed 2020-11-04 by jogibear9988