dotnetprojects/DotNetSiemensPLCToolBoxLibrary #154: Read DB Array Size 1024 with ReadValues not possible

Nick135, 2020-10-31:
Das lesen eines Byte Arrays der Größe 1024 führt bei ReadValues zu einem Exception. Mit ReadValue funktioniert es.

> Operation failed due to error from PLC 255: no message defined!

> at DotNetSiemensPLCToolBoxLibrary.Communication.PLCConnection.ReadValues(IEnumerable`1 valueList, Boolean useReadOptimization) in \DotNetSiemensPLCToolBoxLibrary-master\LibNoDaveConnectionLibrary\Communication\PLCConnection.cs:line 2925

> at DotNetSiemensPLCToolBoxLibrary.Communication.PLCConnection.ReadValues(IEnumerable`1 valueList) in \DotNetSiemensPLCToolBoxLibrary-master\LibNoDaveConnectionLibrary\Communication\PLCConnection.cs:line 2597

	`string ip = "192.168.214.1";

	 string plcAddress = "P#DB123.DBX0.0 Byte 1024";	 // Fehler bei ReadValues
	 //plcAddress = "P#DB123.DBX0.0 Byte 1030";	 	 // Funktioniert
	 using (var myConn = new PLCConnection("Debug"))
	 {
		  myConn.Configuration.CpuIP = ip;
		  myConn.Connect();
		  try
		  {
				var a = myConn.ReadValue(plcAddress);
		  }
		  catch (Exception) { }
		  try
		  {
				var tag = new PLCTag(plcAddress);
				myConn.ReadValue(tag);
		  }
		  catch (Exception) { }
		  try
		  {
				var tags = new List<PLCTag>();
				tags.Add(new PLCTag(plcAddress));
				myConn.ReadValues(tags);
		  }
		  catch (Exception) { }
	 }`

Nick135, 2020-10-31:
Hab den Bug gefunden.

'(maxReadSize - HeaderTagSize)' hinzugefügt.
[image]

Nick135, 2020-10-31:
Was mir noch aufgefallen ist, dass die ReadValue Methode 222 Bytes liest und die ReadValues nur 204 pro PDU.

[image]

jogibear9988, 2020-11-04:
Ich schau es mir an, kann aber bissl dauern.
Problem, wir nutzen meine lib Produktiv, und das ist für mich die kritischste stelle... wenn da was falsch ist stimmen alle werte nicht.
Muss n test für den Fehler machen und prüfen ob sich eine Änderung noch anders auswirkt

jogibear9988, 2020-12-13:
Ja, redvalues nutzt einen anderen request mit einem größeren header, die kann dann nur weniger lesen.

jogibear9988, 2020-12-13:
danke für den hinweis, habs gefixt

commit 874a7c7b (2020-12-13): bugfix reading from plc, fixes #154
closed 2020-12-13 by jogibear9988