dotnetprojects/DotNetSiemensPLCToolBoxLibrary #189: Extract hardware configuration of S7 project

hachzoli, 2023-05-19:
Hello , 

It's any chance to extract informations of all hardware configuration from an S7 project or from config file exported from hardware ?

For example to extract info for module type , allocated inputs/outputs for card , order number of the card , etc....

Thanks.

jogibear9988, 2023-05-19:
For sure, you need to analyze the dbf files, and extend the project to support it. Its much work

StefanHasensperling, 2023-05-19:
Actually i have done some of that work, but did not yet have time to finish and make a pull request.

I managed to extract the content os SDB0 from the project and parse its content.
This basically contains all the configuration from the "Cpu parameters" page in simatic manager.
Clock memory, stack space, io size, remanent memory and such.
I know how to parse that info, but did not yet finish the implementation of all properties.

Will make a pull request with what i have, and someone can go from there.

[X]<http://www.mlogics.cl/>             Stefan Hasensperling
Ingeniero de Automatización
Gerente
MLogics Chile Ltda.
MLogics Chile Ltda.
RUT:  76239183 -K
Monjitas 744, Dep. 409
Santiago
Chile
www.mlogics.cl<http://www.mlogics.cl/>          "El Sistema de Control BatchXpert es Potente, Eficiente y Competitivo, ideal para la automatización de procesos en la industria."
Para conocer más visita nuestra web y solicita el demo en www.mlogics.cl<http://www.mlogics.cl/>

hachzoli, 2023-05-20:
Thanks for suggestion and for sharing your work…..it is a good starting point and for sure will be hard work to continue this.

Unfortunately I am not to familiar with dbf files and I need to understand how data it is bind.

I will try to continue with this project.

jogibear9988, 2023-05-20:
dbf are database files, there are a few opensource prigeams to read them. in in this library also dbf reading is implemented. problem is not dbf reading, it's the data in the dbfs. this is not publicly documented anywhere.
and there are lots of database files in the s7 project

jogibear9988, 2023-05-20:
here is some documentation:

https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/tree/master/LibNoDaveConnectionLibrary/Documentation

and if course some info is only in the code

jogibear9988, 2024-08-25:
I've added the Code form @StefanHasensperling to the Library, see https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/commit/80331eb6f14be126b15e0d5d09bde78d9b993e49

Thx to @StefanHasensperling

closed 2024-08-25 by jogibear9988