dotnetprojects/DotNetSiemensPLCToolBoxLibrary #199: NC Var for Sinumerik One

diegoErol, 2024-01-26:
For Sinumerik One the PLC associated is a S7-1200 comparing with the S7-300 that Sinumerik 840D has. NC Tag read works sometimes for One but suddenly throws an exception if the polling is faster than 3 seconds approx. Maybe, this problem is connected with the issue regarding to PUT/GET operations and DB optimized memory.

rferreiraperez, 2024-01-30:
The same thing happens to me but I don't know where the problem lies, any solution?

Nick135, 2024-02-03:
@diegoErol The SINUMERIK ONE use a S7-1500. The NCK is almost the same so far. (4.9x <-> 6.xx)

You write that you have problems with NC-VAR. This has nothing to do with the PLC.
Communication continues to work normally for me.

What error are you getting exactly?

diegoErol, 2024-02-05:
TheNCK is a 6FC5317.
What I'm experiencing is that one of each two reads is returning a null value. For example if I define a poll time of 10 seconds:

```
Pos:3,02785 at time: 00:00:00.0274315   
Pos:null at time: 00:00:05.0086867    
Pos:3,02784 at time: 00:00:00.0132940    
Pos:null at time: 00:00:05.0075245
```

The response is the same as if define 1 second of pooling time:

```
Pos:3,02785 at time: 00:00:00.0294128
Pos:null at time: 00:00:05.0086773     
Pos:3,02785 at time: 00:00:00.0171700  
Pos:null at time: 00:00:05.0075623
```

If I let run it sequenced:

```
Pos:3,02785 at time: 00:00:00.0289626
Pos:null at time: 00:00:05.0034212
Pos:3,02785 at time: 00:00:00.0157251
Pos:null at time: 00:00:05.0002274
```

It seems like the second read is lasting always the same time and return a null value.

Nick135, 2024-02-07:
I also have this problem with the current NCK version.
I will do some tests to see if the problem is on our side or again in the Siemens firmware.

Nick135, 2024-02-07:
Looks like the NC close the connection after each read request.

Nick135, 2024-02-09:
I have been in contact with Siemens and they have confirmed that it is a fault. I have been given a workaround, which I want to test. At the same time, Siemens wants to investigate the problem further and release a hotfix.

Nick135, 2024-02-29:
Siemens will fix this Bug with CNC Version 6.22 HF3

diegoErol, 2024-02-29:
Very good news! Will try when released

d0pam1n, 2024-03-13:
Same here ✋ 

Anyone already tried with new firmware version?

Nick135, 2024-06-11:
I tried the new firmware CNC Version 6.22 HF3 and can confirm that the bug is fixed.

jogibear9988, 2024-06-11:
Then I close this.

closed 2024-06-11 by jogibear9988