dotnetprojects/DotNetSiemensPLCToolBoxLibrary #207: How can I read the Profibus DP diagnostic data with SFC13 DPNRMDG ?

mut9bu, 2024-08-05:
Hello
If I understood correctly it is possible to read data from a S7-300 PLC  with a system function (SFC).
Is there an example how to do this? In my usecase I want to read DP diagnostic data from Profibus DP slaves that are connected to S7-300 PLC. Additionally: Is there a general documentation that discribes how to connect (disconnect) to (from) a PLC?

jogibear9988, 2024-08-05:
You can not call SFC's with the communication library, the SFC has to be called from a PLC Block.

But you can read SZL's, they content of the SZLs are described here: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/blob/master/LibNoDaveConnectionLibrary/Documentation/System%20Software%20for%20S7-300%20and%20S7-400%20System%20and%20Standard%20Functions.pdf

Look in Apendix B, maybe there is one with the info you need. (maybe B.53)

[image]

And no, there is no better doc, you need to look into the samples.
(this should be the easyest one: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/tree/master/SimpleCSharpDemonstartion)

mut9bu, 2024-08-05:
Hi @jogibear9988 
Thank you for the quick answer. Yes you are right, first I tried it over SZL Id 0x00B4. My target is to read the mentioned data from 840D powerline/solutionline CNC controllers that have inegrated S7-300 CPU's. I thought it should be the same like simple S7-300 PLC's. But I noticed that reading SZL lists from 840D does not give the same results like simple S7-300's. For being sure that the difference is not coming from the library, I have read the SZL's directly from the PLC program of 840D with SFC51 RDSYSST, but result was the same. 
Afterwards I contacted Siemens Support and they told me that it is possible that 840D's S7-300 can differ from normal S7-300 CPU's and suggested me to read the info over SFC13. Therefore I asked whether it's possible to read data over SFC13.
Bu nevertheless I want to have a look at the pdf you have sent me, but If I try to open it I get an Error.
"Error rendering embedded code Invalid PDF"

Do you have any information about the differences of SZL's 840D's S7-300 vs normal S7-300?

mut9bu, 2024-08-05:
> You can not call SFC's with the communication library, the SFC has to be called from a PLC Block.
> 
> But you can read SZL's, they content of the SZLs are described here: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/blob/master/LibNoDaveConnectionLibrary/Documentation/System%20Software%20for%20S7-300%20and%20S7-400%20System%20and%20Standard%20Functions.pdf
> 
> Look in Apendix B, maybe there is one with the info you need. (maybe B.53)
> <img alt="image" width="630" src="https://private-user-images.githubusercontent.com/364896/355108977-85317e32-3223-4a56-bcfc-aed8008f7d35.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NjM5NjYsIm5iZiI6MTcyMjg2MzY2NiwicGF0aCI6Ii8zNjQ4OTYvMzU1MTA4OTc3LTg1MzE3ZTMyLTMyMjMtNGE1Ni1iY2ZjLWFlZDgwMDhmN2QzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQxMzE0MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZTYzNTFiYzlhNGUyMDJiMDcwZDcyYTFiYjY4ZGY2NmE1YmM1YWM4NDVlOTc0NmRkOTExNjUxN2U5OGExNWRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.rNmpQXRkHrEHT6UPLYmVXrMImUSqzeVGDJkFuBTehnw">
> 
> And no, there is no better doc, you need to look into the samples. (this should be the easyest one: https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary/tree/master/SimpleCSharpDemonstartion)

Hi @jogibear9988 
Thank you for the quick answer. Yes you are right, first I tried it over SZL Id 0x00B4. My target is to read the mentioned data from 840D powerline/solutionline CNC controllers that have inegrated S7-300 CPU's. I thought it should be the same like simple S7-300 PLC's. But I noticed that reading SZL lists from 840D does not give the same results like simple S7-300's. For being sure that the difference is not coming from the library, I have read the SZL's directly from the PLC program of 840D with SFC51 RDSYSST, but result was the same. 
Afterwards I contacted Siemens Support and they told me that it is possible that 840D's S7-300 can differ from normal S7-300 CPU's and suggested me to read the info over SFC13. Therefore I asked whether it's possible to read data over SFC13.
Bu nevertheless I want to have a look at the pdf you have sent me, but If I try to open it I get an Error.
"Error rendering embedded code Invalid PDF"

Do you have any information about the differences of SZL's 840D's S7-300 vs normal S7-300?

jogibear9988, 2024-08-05:
No, i never communicated with a CNC