dotnetprojects/DotNetSiemensPLCToolBoxLibrary #210: Is it possible to read  out the diagnostic addresses of the Profibus DP slaves with your library?

mut9bu, 2024-08-12:
Hello

Finally I was able to read out the DP diagnostic data form the DP slaves of a S7-300 system with the System status lists. (SZL ID: 0x00B4  Index: Diagn.Addr.of the related slave). But here I have one problem:
My application is connecting to one of the selected S7-300 PLC in our plant and reading out some diagnostic data. One of them is the DP diagnostic data of the Profibs DP slaves.  I can use the SZL (Sytemzustandsliste) with SZL-ID 0x00B4 for it but only if I know the diagnostic addresses of the slaves. I can get the information only from the S7 Projekt-Hardware Configuration.  But my app should be able to list all the existing Profibus-DP slave diagnostic data.  That means I should be able to read out first all the diagnostic addresses of the DP slaves, and then the related diagnostic data over SZL ID 0x00B4.

Is it possible to read out , all the diagnostic addresses of the existing slaves? I have looked to the supportes SZL-functions but I couldn't find any output like that. Is it perhaps possible to get the data over the online SDB 1XXX Block in the CPU? Or is there another simpler way you know?

jogibear9988, 2024-08-13:
I don't know if it is possible. Think you should ask Siemens this, or ask at the siemens forums or sps-forum.de

closed 2024-08-25 by jogibear9988

mut9bu, 2024-08-29:
Just for information: I have asked the topic in Siemens Automation Forum but didn't get any answer. Also asked Siemens Hotline Support and got following answer:

_eine solche Massenoperation gibt es nicht.
Auch sonst kenne ich keine Operation die ihnen explizit ein Diagnoseadresse zurückliefet.

Eine Möglichkeit, die sie testen sollten, ist der GEO_LOG
Wenn sie hier eine geografische Adresse eingeben, die auf den Slot 0 zeigt, könnten sie die Diagnose Adresse zurückbekommen.

Das ist die einzige Idee dafür die ich habe._