dotnetprojects/DotNetSiemensPLCToolBoxLibrary #216: Is it possible to read *.PG files from the S7-Project (in SDB folder) with DotNetSiemensPLCToolBoxLibrary ?

mut9bu, 2024-09-10:
Hello I have seen that it is possible parse DBF files with DotNetSiemensPLCToolBoxLibrary . But in my case I need to parse data from a *.PG file in the S7 Project. Is that possible? For testing I have tried to open the PG file with a online DBF reader but without succes. But when I opened the *.PG file with the online FILExt app, I could see the content without any problem.
Does anyone know how the *.PG file is encoded and if its possible to read it with DotNetSiemensPLCToolBoxLibrary?

jogibear9988, 2024-09-10:
What information is inside of the PG files?

jogibear9988, 2024-09-10:
I don't know what the contes of those files is, you need to analyze them.
But if they are from a third party software, they have nothing to do with this library

mut9bu, 2024-09-10:
Hello, for correct understanding: PG files are part of the S7-Project structure, not of the 3. part library:

jogibear9988, 2024-09-10:
Here are also some infos about the SDBs....

https://www.sps-forum.de/threads/liste-von-systembausteinen-welcher-ist-f%C3%BCr-was-zust%C3%A4ndig.86031/#post-647381

https://www.sps-forum.de/threads/mc7-code-in-awl-umzuwandeln.27703/page-4#post-606043

https://www.sps-forum.de/threads/globaldaten-auslesen.41518/#post-301946

closed 2024-09-10 by mut9bu

jogibear9988, 2024-09-11:
@mut9bu 
ask process informatik, maybe they can help you.

mut9bu, 2024-09-13:
> And one more, you could ask "Deltalogic" for help. Years ago they had their own Programming Software for Siemens PLCs, wich they stopped developing. Maybe they share some information (maybe for some money).
> 
> And also https://www.mhj-online.de/ has a step7 software, so they also need to understand all of the files.
> 
> And one more: https://www.process-informatik.de/ has also a S7 programming software, so they also know the files

Thank you so much Jochen! I will follow your suggestioons 👍

jogibear9988, 2024-09-13:
If you get some information, it would be nice if you add it to the toolbox (or at least leave some informations)

mut9bu, 2024-09-13:
> If you get some information, it would be nice if you add it to the toolbox (or at least leave some informations)

Definitly, I will do it, thank you again 👍